gbaman/RaspberryPi-LTSP #43: Sonic Pi not launching

vlee489, 2015-01-07:
When I launch Sonic Pi nothing happen and when in the *sonic-pi* command in the terminal it says that the file doesn't exist.

I am using the newest version of RaspberryPi-ltsp.

gbaman, 2015-01-07:
Have you done an ```Update-All``` any time recently? Goes through the full install list so should sort it out unless there is an issue with Sonic-pi itself

vlee489, 2015-01-07:
If that's the thing it asks you to do after a update then yes

gbaman, 2015-01-07:
Weird, try 
```sudo ltsp-chroot --arch armhf dpkg-reconfigure sonic-pi```
Will reconfigure the package, if you get an error about Sonic-pi not being installed, try these 2 commands
```sudo ltsp-chroot --arch armhf apt-get purge sonic-pi```
```sudo ltsp-chroot --arch armhf apt-get install sonic-pi```

vlee489, 2015-01-07:
On the server or the pi's?

gbaman, 2015-01-07:
Everything is done on the server.
```sudo ltsp-chroot --arch armhf``` means run command in the virtual Pi OS. 
http://pi-ltsp.net/advanced/editing-outside.html

vlee489, 2015-01-07:
Thanks I'll try that when I get in ti school tomorrow

vlee489, 2015-01-12:
We have found the problem. Currently it isn't a problem on Raspberry pi LTSP but it is that but the foundation haven't realsed the software to see how to fix it.

ryanteck, 2015-01-18:
Temporary fix can be done by installing the package

libqt4-network

gbaman, 2015-01-18:
Can confirm this fixes the issue

gbaman, 2015-01-18:
Can confirm a bugfix workaround has now been added in Release 0.10.39 https://github.com/gbaman/RaspberryPi-LTSP/commit/aa27c90312ed42930e1bffd188035a021d09f3e3
Just upgrade your Raspi-LTSP and make sure to accept the packages update.

closed 2015-01-18 by gbaman