gbaman/RaspberryPi-LTSP #44: cannot use keyboard and mouse not working once login screen is display

wegunterjr, 2015-01-08:
just updated today to 10.3.8 and I have 3 Pi's.  They loaded quickly to the boot screen and recognized the keyboards and mice, but once I got to the login screen, i was unable to use the keyboard or mouse. The mouse had a light on underneath, but the keyboard wouldn't show any indication of power or operation.   Any thoughts?

gbaman, 2015-01-08:
This is due to a kernel update, although the update should have been applied to your SD cards automatically with the new kernel updater added in 0.10.38..
Can you post the output of
```ls /opt/ltsp/armhf/etc/init.d```
and
```ls/opt/ltsp/armhf/etc/rc2.d```

The main thing to verify is is there a file called ```kernelCheckUpdate.sh``` in both? The rc2.d command may have a slightly different named one, but name will contain ```kernelCheckUpdate.sh```.

wegunterjr, 2015-01-08:
The email seems to have been cut because I only see the first part:
ls /opt/ltsp/armhf/etc/init.d

kernelCheckUpdate is in there.

Also, it did build the SD card when i did the update, so i just did it
again to make sure. It didn't take very long...less than a minute.

Walter

gbaman, 2015-01-13:
Any luck @wegunterjr?
I need to see the results of ```ls /opt/ltsp/armhf/etc/rc2.d``` if you are still having this issue.

wegunterjr, 2015-01-13:
> On Tue Jan 13 2015 at 12:53:46 PM Andrew Mulholland <

Andrew,
 Sorry for the delay.  I was able to re-run the update to 10.38, and it
deployed the SD Card files.  Once I did that, they booted up and so far
allowed for use of keyboard and mouse.
Thanks for the follow up. My kids were able and still continue to do
Scratch and Minecraft (I have 3 pi's and they can all login to the same
world and collaborate or destroy). AWESOME!!!

Walter

gbaman, 2015-01-14:
Great to hear it is all working now.

closed 2015-01-14 by gbaman