gbaman/RaspberryPi-LTSP #45: New Raspberry Pi UI

liamn, 2015-01-21:
Is it possible to get the new Raspberry Pi UI that was released over Christmas working with Pi-LTSP? Would running the distribution upgrade script inside a chroot be suitable?

Thanks,
Liam

gbaman, 2015-01-21:
I am in the process of getting it working.
It included a number of what seems to be possible "shortcuts" taken so is causing me a few issues making everything work.
Will try and get it sorted tonight.

You can just manually install as per the Raspberry Pi foundation instructions, it but it has a few issues including no stary desktop background yet etc.
Also, be warned, if you already have it installed, when I release the upgrade for Raspi-LTSP, it may not detect an upgrade is required for that feature.

liamn, 2015-01-21:
Working fine here! Just a missing raspberry pi logo on the menu button, but i'm guessing that's a missing file. The desktop wallpaper was just a case of reapplying it through the inbuilt settings. 

Thanks for the help!

gbaman, 2015-02-04:
New UI stuff finally added to main Raspi-LTSP in Release 0.10.41 https://github.com/gbaman/RaspberryPi-LTSP/commit/b333306ed49962985471546f7de001da1b0a42ef

closed 2015-02-04 by gbaman