gbaman/RaspberryPi-LTSP #46: Enable ability to install a DHCP server.

ryanteck, 2015-01-28:
Would be nice to have the ability to install a DHCP server on the server machine.

When trying to do it didn't work so would need experimenting with IRC.

william1616, 2015-01-28:
its entirely possible i've done it with both isc-dhcp-server and dnsmasq

Do you get any error messages? If you do not you probably need to add 
```
INTERFACES="eth0"
```
to 
```
/etc/default/dhcp3-server
```

http://chschneider.eu/linux/server/dhcp3.shtml is the instructions that I followed to achieve this

ryanteck, 2015-01-28:
I can't remember what the error was actually. I couldnt spend much time on it.

Going to use another pi as the server as I can tweak it to work out of class without breaking anything. Also provides a nice Wi-Fi bridge for internet access.

closed 2015-01-28 by ryanteck