gbaman/RaspberryPi-LTSP #47: Feature Request for Raspberry Pi 2 Compatibility

michaelgibson, 2015-02-28:
Requesting Pi 2 support!
I'm working on a project that requires the new capabilities of the pi 2's and would love to be able to use this utility.
I understand from the FAQ that the pi 2's are not currently supported(as of Feb 5, 2015).
http://pi-ltsp.net/faq.html

Let me know what would be involved for getting this functioning with them and maybe I could help?
Thanks!

gbaman, 2015-03-02:
It is coming... Slowly. There have been a number of hurdles so far including.
- Waiting for Raspbian team kernels (more explanation of the different kernels here - http://pi-ltsp.net/advanced/kernels.html).
- AUFS support has been dropped in the next kernels.
- OverlayFS support (which we can use instead of AUFS) has been added.
- OverlayFS was renamed to just Overlay in the kernel modules.

I am currently at the last issue, as LTSP (base package under Raspi-LTSP) expects overlayfs modul to be called just that, but it is called Overlay, which has been causing issues.

Plus, on top of all of this, I have been pretty busy outside of the project. Don't forget, if you need it urgently, you can build your own http://pi-ltsp.net/advanced/kernels.html
Hopefully when I get more time to debug even more, it will come officially...

michaelgibson, 2015-03-03:
Ok good to know.
I'll have a look more at the build docs and see where that takes me.
Thanks!

gbaman, 2015-03-16:
Added
https://github.com/gbaman/RaspberryPi-LTSP/commit/80c2f42eed384b46829ac27f83103340577e93b1

closed 2015-03-16 by gbaman