gbaman/RaspberryPi-LTSP #49: Boot errors

developius, 2015-03-07:
I'm having very strange boot issues...
[image]

gbaman, 2015-03-07:
That is an interesting one.. Not seen it get that far before. Can you verify on the SD card, in the cmdline.txt file that the IP address following ```nbdroot=``` is the same IP address displayed at the top right of Raspi-LTSP control panel?
Can I also check, you are connecting the Raspberry Pi and the server to normal a network and not just directly between each other?

developius, 2015-03-07:
IP is the same and yes, I'm connecting accross a network :)

gbaman, 2015-03-07:
Can you try running ```Refresh System``` in the ```Other``` menu of Raspi-LTSP control panel?
Can I also confirm you are using a Raspberry Pi 1 model B or Raspberry Pi 1 model B+?

developius, 2015-03-07:
Refreshing didn't help :( yup, RPi 1 model B.
Can I just ask what the prompt is I'm getting to, it seems very basic?

developius, 2015-03-07:
Sorry - didn't mean to close!

gbaman, 2015-03-07:
Hmm, you have me baffled... Yeah, it is the initramfs prompt. It is a very limited busybox based ash (similar to bash, just older with less stuff) shell. It is running off the tiny operating system stored on the SD card used to connect to the main one on the server.
It seems to be complaining the virtual ram for the Pi (which is stored on the server) is failing to be allocated correctly.
Then it goes on to complain the AUFS filesystem mount is failing to correctly mount the root filesystem.
AUFS works by mounting the master operating system image (which it can't find) then layers a second operating system on top. If you edit a file, it gets saved on the upper layer and put over the top of the read only one on the bottom. It is actually stored away somewhere else. So you only see the new edited version, while the master one is untouched.
System usually works great, but of course if you can't find the bottom OS, you are kind of stuck.

Am not sure what to recommend now..  The version you are running is a pretty stable version, it was released start of February. There have been no other bug reports like this.
Did you install it on a virtual machine or is it running on an old computer? Remember, if installing it on an old computer, it should have been a fresh reinstall from scratch.

developius, 2015-03-07:
Thanks for the info about the prompt :). It makes sense, but I'm not sure why that would be happening considering both of the are networked fine (have tried both ethernet cables on my laptop). I have installed Ubuntu on an old computer, which *does* have very limited RAM, but surely everything is running on the Pi and it only grabs it's OS from the server?
I did have a few problems formatting the SD card - although I *think* it's ok, or it wouldn't boot(?).

gbaman, 2015-03-07:
Limited memory may be the issue, giving it is failing at the virtual memory reservation. How much is limited?
Also, to check, do a ``df -h`` in a terminal on the server, just to check there is actually disk space on the main partiton (```/dev/sdX```).
Also, if your card isn't formatted correctly, that could cause issues.

Although I have an idea. On the SD card, delete the ```config.txt``` file and replace it with ```config_standard.txt```.
The main difference is the normal ```config.txt``` file has overclocking turned on which some old Pis don't like much. It is being disabled in the next kernel file release.

developius, 2015-03-07:
It has 512MB RAM :/ and I've got 31GB disk space available.
I just did that, but it seems they both have overclocking enabled?

gbaman, 2015-03-16:
@developius can you drop me an email, I have an idea.
http://pi-ltsp.net/support.html

closed 2015-03-21 by gbaman