gbaman/RaspberryPi-LTSP #50: missing python3-feedparser dependency

gryphius, 2015-03-09:
in the latest update  0.10.42  , pi-ltsp requires the python feedparser module, which isn't installed by default

```
sudo bash Pi_ltsp
Traceback (most recent call last):
  File "/usr/local/bin/Pi_ltsp-functions-python.py", line 494, in <module>
    checkUpdate(sys.argv[2])
  File "/usr/local/bin/Pi_ltsp-functions-python.py", line 371, in checkUpdate
    import feedparser
ImportError: No module named 'feedparser'
```
workaround:

```
sudo apt-get install python3-feedparser -y
```

gbaman, 2015-03-09:
Can I confirm you are running standard Ubuntu 14.04 as recommended in the documentation?
Please keep in mind stock Ubuntu 14.04 is the only supported operating system for Raspi-LTSP right now.
In saying that, there isn't an issue in explicitly including it.

gryphius, 2015-03-09:
argh.. sorry, this is on a ubuntu server edition. *you shall not open github issues before your first coffee in the morning* - closing and running back to my cave.

closed 2015-03-09 by gryphius

gbaman, 2015-03-09:
Yeah, I don't recommend the server edition given the software is targeted for teachers who just a raw commandline could be an issue for.
In saying that, pushed a fix anyway as the auto updater being broke on any platform can't be a good thing.
I will though have to add some detection mechanism for people running Ubuntu server to warn them I don't test with it.
https://github.com/gbaman/RaspberryPi-LTSP/commit/c5170bc03dc745acf82b6ae91daeda508cb48197