gbaman/RaspberryPi-LTSP #52: wireless raspberry pi classroom

fabubmz07, 2017-12-21:
is it possible to connect a pinet server wirelessly to a raspberry pi? I am currently working on a raspberry pi wireless classroom project (server client based).
students user accounts are stored on the central pinet server with automated backup.

fabubmz07, 2017-12-21:
i just wanted to connect 2 pis and one main server with a single wireless connection

gbaman, 2017-12-21:
This repository is no longer maintained, the PiNet repository can be found over at https://github.com/pinet/pinet

As is noted in many places on the PiNet website, no PiNet does not support booting over wifi. For more information, please read the [PiNet FAQ](http://pinet.org.uk/articles/faq.html).

closed 2017-12-21 by gbaman